dotnet-bluetooth-le/dotnet-bluetooth-le #933: iOS adapter.ConnectToDeviceAsync is very very slow

ideveloperaz, 2025-03-07:
Hi everyone

iOS adapter.ConnectToDeviceAsync is very very slow ... it takes few min when it could connect to device. What to do? how to improve?
Android can connect within few sec

axa88, 2025-03-09:
> Hi everyone
Hi

> 
> iOS adapter.ConnectToDeviceAsync is very very slow ... it takes few min when it could connect to device.

Idk maybe it's tripping up like window can #921

> What to do? how to improve?

Could start by providing way more info than you have, particular devices used is probably most relaxant..

> Android can connect within few sec

Ya Android is really the only first class citizen of this lib.

MahdiGhiasi, 2025-04-06:
The same thing happens on Windows 11 as well.

I'm testing on Surface Pro 11, connecting to a Gatt server on Nothing Phone 2a.

This code takes forever:

```
await CrossBluetoothLE.Current.Adapter.ConnectToDeviceAsync(device, new Plugin.BLE.Abstractions.ConnectParameters(), cancellationTokenSource.Token);
```

But just bypassing the cross-platform routing of the library and directly calling the Windows specific implementation of this very same library works fine and connects quickly:

```
var deviceWin = device as Plugin.BLE.Windows.Device;
var result = await deviceWin.ConnectInternal(new Plugin.BLE.Abstractions.ConnectParameters(), cancellationToken: cancellationTokenSource.Token);
```