dotnet-bluetooth-le/dotnet-bluetooth-le #939: Android project reference could not be resolved

greatPHPer, 2025-04-11:
## Steps to reproduce

1.Add MvvmCross.Plubin.BLE version 3.1.0 to Xamarin Forms project (Xamarin forms Version 5.0.0.2196)

2.Debug it in Android emulator

## Expected behavior
It should run without any bugs

## Actual behavior
It gives a bug in Android project
Can not resolve reference: Plugin.BLE.Abstractions, referenced by . Please add a NuGet package or assembly reference for `Plugin.BLE.Abstractions`, or remove the reference to .

### Crashlog

1>Starting deployment to pixel_7_-_api_33 ...
1>Deploying to pixel_7_-_api_33 ...
Build started.
Project "TOR_tra_actual_forms.Android.csproj" (Install target(s)):
Found Java SDK version 17.0.14.
Found Java SDK version 17.0.14.
Dex Fast Deployment Enabled: False
MonoAndroid Tools: E:\Program Files\Microsoft Visual Studio\2022\Enterprise\MSBuild\Xamarin\Android\
Android Platform API level: 33
TargetFrameworkVersion: v13.0
Android NDK: 
Android SDK: E:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\SHARED\ANDROID\ANDROID-SDK\
Android SDK Build Tools: E:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\SHARED\ANDROID\ANDROID-SDK\build-tools\32.0.0\
Java SDK: C:\Program Files\Microsoft\jdk-17.0.14.7-hotspot\
Application Java class: android.app.Application
Project "TOR_tra_actual_forms.csproj" (GetTargetFrameworks target(s)):
Done building project "TOR_tra_actual_forms.csproj".
Project "TOR_tra_actual_forms.csproj" (GetTargetPath target(s)):
Done building project "TOR_tra_actual_forms.csproj".
Project "TOR_tra_actual_forms.csproj" (GetNativeManifest target(s)):
Done building project "TOR_tra_actual_forms.csproj".
Found conflicts between different versions of "System.Numerics.Vectors" that could not be resolved.
There was a conflict between "System.Numerics.Vectors, Version=2.0.5.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a" and "System.Numerics.Vectors, Version=4.1.3.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a".
    "System.Numerics.Vectors, Version=2.0.5.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a" was chosen because it was primary and "System.Numerics.Vectors, Version=4.1.3.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a" was not.
    References which depend on "System.Numerics.Vectors, Version=2.0.5.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a" [E:\Program Files\Microsoft Visual Studio\2022\Enterprise\Common7\IDE\ReferenceAssemblies\Microsoft\Framework\MonoAndroid\v1.0\System.Numerics.Vectors.dll].
        E:\Program Files\Microsoft Visual Studio\2022\Enterprise\Common7\IDE\ReferenceAssemblies\Microsoft\Framework\MonoAndroid\v1.0\System.Numerics.Vectors.dll
          Project file item includes which caused reference "E:\Program Files\Microsoft Visual Studio\2022\Enterprise\Common7\IDE\ReferenceAssemblies\Microsoft\Framework\MonoAndroid\v1.0\System.Numerics.Vectors.dll".
            System.Numerics.Vectors
    References which depend on or have been unified to "System.Numerics.Vectors, Version=4.1.3.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a" [].
        C:\Users\DHILIPKUMARR\.nuget\packages\system.text.json\8.0.5\lib\netstandard2.0\System.Text.Json.dll
          Project file item includes which caused reference "C:\Users\DHILIPKUMARR\.nuget\packages\system.text.json\8.0.5\lib\netstandard2.0\System.Text.Json.dll".
            C:\Users\DHILIPKUMARR\.nuget\packages\system.text.json\8.0.5\lib\netstandard2.0\System.Text.Json.dll
            C:\Users\DHILIPKUMARR\.nuget\packages\refit\8.0.0\lib\netstandard2.0\Refit.dll
            E:\Users\DHILIPKUMARR\source\repos\TOR_tra_actual_forms\TOR_tra_actual_forms\TOR_tra_actual_forms\bin\Debug\netstandard2.0\TOR_tra_actual_forms.dll
            C:\Users\DHILIPKUMARR\.nuget\packages\system.net.http.json\8.0.1\lib\netstandard2.0\System.Net.Http.Json.dll
Skipping target "_ResolveLibraryProjectImports" because all output files are up-to-date with respect to the input files.
Skipping target "_BuildLibraryImportsCache" because all output files are up-to-date with respect to the input files.
Skipping target "_GenerateAndroidResourceDir" because all output files are up-to-date with respect to the input files.
Skipping target "_GenerateLayoutBindings" because it has no outputs.
Skipping target "_ConvertResourcesCases" because all output files are up-to-date with respect to the input files.
Skipping target "_CompileResources" because all output files are up-to-date with respect to the input files.
Skipping target "_PrepareUpdateAndroidResgen" because all output files are up-to-date with respect to the input files.
Skipping target "_UpdateAndroidResgen" because all output files are up-to-date with respect to the input files.
  Adding assembly reference for TOR_tra_actual_forms.Android, recursively...
  Adding assembly reference for netstandard, recursively...
    Adding assembly reference for System.Data, recursively...
      Adding assembly reference for System.Transactions, recursively...
    Adding assembly reference for System.Data.DataSetExtensions, recursively...
    Adding assembly reference for System.IO.Compression, recursively...
    Adding assembly reference for System.IO.Compression.FileSystem, recursively...
    Adding assembly reference for System.ComponentModel.Composition, recursively...
    Adding assembly reference for System.Web.Services, recursively...
  Adding assembly reference for System.Drawing.Common, recursively...
  Adding assembly reference for System.Net.Http, recursively...
  Adding assembly reference for System.Runtime.Serialization, recursively...
[... 85 lines cut ...]
    Adding assembly reference for Xamarin.AndroidX.VectorDrawable, recursively...
    Adding assembly reference for Xamarin.AndroidX.VersionedParcelable, recursively...
    Adding assembly reference for Xamarin.AndroidX.ViewPager, recursively...
    Adding assembly reference for Xamarin.AndroidX.ViewPager2, recursively...
    Adding assembly reference for Xamarin.Essentials, recursively...
    Adding assembly reference for Xamarin.Forms.Core, recursively...
    Adding assembly reference for Xamarin.Forms.DataGrid, recursively...
    Adding assembly reference for Xamarin.Forms.Platform.Android, recursively...
    Adding assembly reference for Xamarin.Forms.Platform, recursively...
    Adding assembly reference for Xamarin.Forms.Xaml, recursively...
    Adding assembly reference for Xamarin.Google.Android.Material, recursively...
    Adding assembly reference for Xamarin.Google.Guava.ListenableFuture, recursively...
Done building project "TOR_tra_actual_forms.Android.csproj" -- FAILED.
Build FAILED.
1>Deployment failed to pixel_7_-_api_33.
========== Deploy: 0 succeeded, 1 failed, 0 skipped ==========
========== Deploy completed at 12:48 and took 01.175 seconds ==========

## Configuration

**Version of the Plugin:** 3.1.0

**Platform:** Xamarin Forms 5.0.0.2196

**Device:** Android SDK Emulator Pixel 7

greatPHPer, 2025-04-11:
I changed the version to 2.1.3 in android project also. It was different and it was 3.1.0
Now it is debugging in android emulator

closed 2025-04-11 by greatPHPer

janusw, 2025-04-11:
Version 3.1.0 primarily targets .NET 7 and 8. Technically it does still include the old bindings for netstandard2.0 and Xamarin, but I'm not sure anyone is still seriously using them.

If you're still using Xamarin at this point, I'd strongly recommend to migrate to MAUI (or some alternative framework). Xamarin is badly of out support for a while (almost a year now). MAUI 7 was pretty bad in terms of quality IMHO, but MAUI 8 and 9 should be stable enough to replace Xamarin. Future releases of this library will not care about compatibility with Xamarin any more.